MicrosoftDocs/windows-driver-docs-ddi #4: some dead title link,could you fix?

nanji2018, 2018-01-25:
hello:

1.i found a dead link in this page.
https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/_sensors/..%5Cgnssdriver%5Cindex.md

2.i want to download Gnssdriver.h file,could i download them in github?like below path
Windows \HardwareProgramming reference for Windows Driver Kit\Sensors\Gnssdriver.h
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/gnssdriver/ni-gnssdriver-ioctl_gnss_start_fixsession

thanks

tedhudek, 2018-01-25:
@barrygolden please take a look

barrygolden, 2018-01-26:
Investigating broken link.

barrygolden, 2018-01-26:
Hi @nanji2018,
You can find the gnssdriver.h file in the Windows Driver Kit (WDK) at this location:
    C:\Program Files (x86)\Windows Kits\10\Include\10.0.16299.0\shared\gnssdriver.h

If you don't have the WDK, you can download and install it by following the instuctions on this site:
    https://developer.microsoft.com/en-us/windows/hardware/windows-driver-kit

Thanks for bringing our attention to the broken link on the docs site, we'll be fixing this issue soon.

closed 2018-01-26 by barrygolden