MicrosoftDocs/windows-driver-docs-ddi #18: Where can I find the corresponding table of variables and values?

SuperSuperSuperSuper5, 2018-03-02:
Just like the 'NtQueryDirectoryFile' function, here is a 'Parameters' named 'FileInformationClass', 
I used bing search a large number of sites to get the value of 'FileBothDirectoryInformation' is equal to 3, I need this value when I'm debugging windows kernels driver, It is not easy to get this information in this document, so I hope MSDN in the preparation of these documents, the variables and the value of the table can be added to facilitate debugging kernel driver

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: deb5372d-d27e-48a3-1a84-0ba62e549d84
* Version Independent ID: 3b37a61b-c55f-6a45-e941-622884520447
* [Content](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-ntquerydirectoryfile#parameters)
* [Content Source](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-ntquerydirectoryfile.md)
* Service: unspecified

EliotSeattle, 2018-03-02:
Assigning to @PageWriter-MSFT.

PageWriter-MSFT, 2018-03-02:
Hello isinstance

Thank you for the feedback. Unfortunately, at this time we are unable to provide enum values. You can view the values in the debugger by using the **dt** command or view them in the header. 

Apologize for the inconvenience. We'll definitely consider this as a feature request.

Please let us know if you have any other comments! 

Page Writer

closed 2018-03-02 by PageWriter-MSFT

Jayzisinstance, 2018-03-12:
There is still a problem, I can not get the **Windows** header source file because it not open source , and the **dt** command can not find the corresponding value

```
kd> dt FileBothDirectoryInformation
*************************************************************************
***                                                                   ***
***                                                                   ***
***    Either you specified an unqualified symbol, or your debugger   ***
***    doesn't have full symbol information.  Unqualified symbol      ***
***    resolution is turned off by default. Please either specify a   ***
***    fully qualified symbol module!symbolname, or enable resolution ***
***    of unqualified symbols by typing ".symopt- 100". Note that     ***
***    enabling unqualified symbol resolution with network symbol     ***
***    server shares in the symbol path may cause the debugger to     ***
***    appear to hang for long periods of time when an incorrect      ***
***    symbol name is typed or the network symbol server is down.     ***
***                                                                   ***
***    For some commands to work properly, your symbol path           ***
***    must point to .pdb files that have full type information.      ***
***                                                                   ***
***    Certain .pdb files (such as the public OS symbols) do not      ***
***    contain the required information.  Contact the group that      ***
***    provided you with these symbols if you need this command to    ***
***    work.                                                          ***
***                                                                   ***
***    Type referenced: FileBothDirectoryInformation                  ***
***                                                                   ***
*************************************************************************
Symbol FileBothDirectoryInformation not found.
```
I think it's really big problem...

PageWriter-MSFT, 2018-03-12:
Hello,

This documentation set describes APIs that are declared in WDK headers. Please download the WDK from: https://developer.microsoft.com/en-us/windows/hardware/windows-driver-kit.

The enumeration is declared in the header, wdm.h under the km folder. 

For information about the **dt** command, see https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/dt--display-type-?branch=master

Page Writer