MicrosoftDocs/windows-driver-docs-ddi #19: Actual type of dwResult

norepudiation, 2018-03-02:
In this structure dwResult is DWORD, that is , plain win32 error code. But IBidiRequest::GetResult() specifies it as HRESULT, a COM error code. Which one is semantically correct ? 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2d752c7c-6018-3d0b-055e-87a06a0f505d
* Version Independent ID: 05e640d9-06bd-6f00-e681-0b19b985b970
* [Content](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/winspool/ns-winspool-_bidi_response_data)
* [Content Source](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/winspool/ns-winspool-_bidi_response_data.md)
* Service: unspecified

barrygolden, 2018-03-02:
Self-assigned @barrygolden for investigation.

barrygolden, 2018-03-02:
Contacted the print driver feature team for clarification on construct elements in the [_BIDI_RESPONSE_DATA](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/winspool/ns-winspool-_bidi_response_data) structure and the [IBidiRequest::GetResult](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/bidispl/nf-bidispl-ibidirequest-getresult) method.

barrygolden, 2018-04-02:
Opened internal work item [16710618](https://microsoft.visualstudio.com/DefaultCollection/OS/_workitems/edit/16710618) to address this issue. Will republish topic when the changes to clarify error code type have been completed,

closed 2018-04-02 by barrygolden