MicrosoftDocs/windows-driver-docs-ddi #21: what?

ccdriver, 2018-03-05:
RegistryPathName

TBD

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e102f060-73ed-afad-e988-1d935f58ea2e
* Version Independent ID: 57387df2-7f18-2dc1-3ea9-95c96a8af4c3
* [Content](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ks/nf-ks-ksinitializedriver#requirements)
* [Content Source](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ks/nf-ks-ksinitializedriver.md)
* Service: unspecified

barrygolden, 2018-03-05:
Self-assigned @barrygolden for investigation.

barrygolden, 2018-03-05:
The description was inadvertently dropped during migration from MSDN to docs.microsoft.com.

It should read as follows::

_RegistryPath_ 
        A pointer to a Unicode string containing the registry path string passed into the minidriver's **DriverEntry** function by the operating system.

Will update the topic and republish.

barrygolden, 2018-03-05:
Added missing parameter description in PR #22. This will be republished in the next daily update.

closed 2018-03-05 by barrygolden