MicrosoftDocs/windows-driver-docs-ddi #24: tilde is 0x7e (not 0x126)

tandasat, 2018-03-13:
The description for the `Tag` parameter says tilde is 0x126, but it is 126 in decimal and 0x7e in hex.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: edc6540c-b149-ed6a-49b3-25ad080fb719
* Version Independent ID: d6cb600f-2879-8455-43db-d80e664e24e6
* Content: [ExAllocatePoolWithTag function | Microsoft Docs](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-exallocatepoolwithtag)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-exallocatepoolwithtag.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-exallocatepoolwithtag.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-03-13:
Thanks @tandasat I've made the change. Due to ongoing work with our publishing pipeline, it might be a few days before the site picks up the change.  In the future, if you like, you can use the Edit button to submit a PR. Thanks again!

closed 2018-03-13 by tedhudek

tandasat, 2018-03-13:
Thank you for letting me know the tips. I thought I could do a PR but could not figure out how and where this repo was.