MicrosoftDocs/windows-driver-docs-ddi #25: The `Tag` to use for `ExFreePoolWithTag` should be documented

tandasat, 2018-03-13:
This document states that developers can use `ExFreePoolWithTag` to free allocated `TokenInformation` but does not specify what `Tag` value should be used. Because of the fact that `ExFreePool` can also be used, developers would assume zero of `Tag` could be used, but a document of   `ExFreePoolWithTag` explicitly prohibit it (`Each ASCII character in the tag must be a value in the range 0x20 (space) to 0x126 (tilde)`). (In fact, a tag value used for allocation is not zero (`  eS`, specifically), so freeing with zero causes tag leak technically.)

To avoid this confusion with `ExFreePoolWithTag`, this document should be explicit about what `Tag` can or should be used for deallocation.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: aa719a3c-1b44-240a-204e-b07d4568cdd0
* Version Independent ID: a428da50-0d29-7aff-e0c7-20e765b8fad6
* Content: [SeQueryInformationToken function | Microsoft Docs](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-sequeryinformationtoken)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-sequeryinformationtoken.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-sequeryinformationtoken.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

AndrewHarryKim, 2018-04-02:
Made change to https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-exfreepoolwithtag

tandasat, 2018-04-02:
I am bit confused at this change. The issue was that the document of the SeQueryInformationToken function did not state what tag value should be used, and the linked change did not clear that point.

AndrewHarryKim, 2018-04-02:
Rather than changing the API for ExFreePoolWithTag to reflect the range of ExFreePoolWithTag, isn't it more appropriate, to change documentation of ExFreePoolWithTag to address this?

tandasat, 2018-04-02:
The thing is that SeQueryInformationToken will be allocating pool with some tag value that might be zero, `Se  `, or something else, which developers cannot tell as it is not documented. 

The update just made describes the general rule of the accepted tag value. However, as the documentation of ExFreePoolWithTag says
```
Specifies the tag value passed to ExAllocatePoolWithTag or ExAllocatePoolWithQuotaTag when the block of memory was originally allocated.
```
, developers are expected to pass the same tag value for ExAllocatePoolWithTag to free pool allocated by SeQueryInformationToken or the tag leaks. Yet, the tag value to use is not specified, which confuses developers and pretty much forces to use ExAllocatePool knowing it would leak a tag.

Sorry if I am not describing the issue clearly. Hope it makes sense.

AndrewHarryKim, 2018-04-03:
It appears that a developer should use only ExFreePool for the tag. Currently a developer may use ExAllocatePool in lieu of ExFreePool but in the future the behavior of this API may change.

tandasat, 2018-04-03:
Limiting only to ExFreePool for freeing `TokenInformation` makes sense to me.

tandasat, 2018-04-03:
Thanks@AndrewHarryKim