MicrosoftDocs/windows-driver-docs-ddi #29: Does this return the current configuration, or the first configuration?

efalk, 2018-03-22:
Document is unclear about what happens if the device is not in its first or default configuration.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e6444c42-ce07-fb8a-a530-edbc443c144a
* Version Independent ID: c6ddf197-a4c8-03ff-60d4-60bf5bb1a8fb
* Content: [WdfUsbTargetDeviceRetrieveConfigDescriptor function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdfusb/nf-wdfusb-wdfusbtargetdeviceretrieveconfigdescriptor)
* Content Source: [wdk-ddi-src/content/wdfusb/nf-wdfusb-wdfusbtargetdeviceretrieveconfigdescriptor.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfusb/nf-wdfusb-wdfusbtargetdeviceretrieveconfigdescriptor.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-03-22:
@Shyamal-V any guidance on this one?

tedhudek, 2018-07-19:
Hi @efalk sorry for the delay.

To select a different USB configuration for a device (for example if the device is not in its first or default configuration), the driver can call [**WdfUsbTargetDeviceSelectConfig**](https://docs.microsoft.com/windows-hardware/drivers/ddi/content/wdfusb/nf-wdfusb-wdfusbtargetdeviceselectconfig) before calling **WdfUsbTargetDeviceRetrieveConfigDescriptor**.

Hope this helps.  I've added the above note to the page you cited, in the **Remarks** section.  The change will be live later today.

closed 2018-07-19 by tedhudek