MicrosoftDocs/windows-driver-docs-ddi #33: Unclear relations between MiniportFilterResourceRequirements and MiniportStartDevice

I-gor-C, 2018-03-23:
Quoting from https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ndis/nc-ndis-miniport_pnp_irp :

> If a miniport driver adds new resources in the MiniportFilterResourceRequirements function, it should provide a MiniportStartDevice function to remove the resources.

Why would I want to remove the resources that were explicitly requested just a moment earlier, before the device was even initialized?

Reading further:
> If a miniport driver modifies resources in such a way that an underlying bus driver cannot recognize the resources, the driver should provide a MiniportStartDevice function to remove the resources.

So _MiniportStartDevice_ is needed here only to handle the failing scenario? If so, it should be stated clearly. Also, if I'm sure that the resources are recognizable, is there still a need to add _MiniportStartDevice_?

duncanmacmichael, 2018-03-24:
Hi @I-gor-C , good question. 

MiniportStartDevice is needed any time you add resources during MiniportFilterResourceRequirements, whether there is a failure or not. In MiniportStartDevice, you remove any resources you added during MiniportFilterResourceRequirements except for message interrupt resources.

I don't think the first sentence meant to imply that what you requested just a moment earlier will be removed right afterwards.

Does that help? If it does, I can clarify the doc. If it doesn't, can you explain what you're trying to accomplish so I can ask the right questions?

I-gor-C, 2018-03-24:
Hi @duncanmacmichael , thanks for your reply.

There are few things unclear to me:
Driver shouldn't add resources except for interrupts:
> Miniport drivers should not modify other resources, such as CmResourceTypeMemory and CmResourceTypePort resources. Miniport drivers should avoid adding a new resource to the resource list.

So why would I have any other added resources in MiniportAddDevice and why would I want to remove them if I explicitly added them earlier?

duncanmacmichael, 2018-03-26:
Yeah, now that I'm starting to see some of the bigger picture I can see why the remarks of these two callbacks seem to be saying different things. Let me see what I can find out for you.

jtippet, 2018-03-26:
There are 3 kinds of changes you might want to make to the resource list:
1. Adding things
2. Removing things
3. Modifying things

In most cases, an NDIS miniport driver only modifies the processor affinity of interrupt resources (i.e., (3) on the list above).  For that typical case, the miniport driver needs only to provide a MiniportFilterResourceRequirements handler, where it can modify the interrupt vectors that were already assigned to it.

In exotic cases, you may want to add things to the resource list (i.e., (1) on the list above).  E.g., you might "somehow" know that your hardware has an IO port range assigned to it, but for whatever reason, you know that the NT kernel cannot see that.  To add a resource to the list, you need to first add it during MiniportFilterResourceRequirements (as before), but then remove it in MiniportStartDevice.

Why remove the resource shortly after adding it?  Because you are inventing the resources, the underlying bus driver won't know what to do with them.  So you inject the resources just before the NT arbiter looks at them, and then hide them again before the final resource list is passed down to the bus driver.  (To use an NDIS analogy: imagine a LWF that inserts a network header in all packets that go out to the network.  It would need to delete the header before completing the packets back to the protocol, since the protocol doesn't know anything about your header.)

All that said, it's super rare for NDIS NIC drivers to need to inject new resources.  It's so rare that the documentation just says "avoid doing this".  In most cases, someone who is tempted to add a new resource is probably solving the wrong problem.  Most of these artificial resources should be added by ACPI tables or other features, rather than having the FDO inject them itself.

Incidentally, it's not possible to correctly *delete* a resource from the list from an NDIS miniport driver (i.e., (2) in the list above); NDIS simply doesn't provide a callback for the downcall of IRP_MN_FILTER_RESOURCE_REQUIREMENTS.  If you really happen to need that, you can ship an NT upper filter driver that does this.

I-gor-C, 2018-03-27:
Hi @jtippet , thanks for the elaborated answer. Just a couple of clarifications:
- Let's say I add a resource in _MiniportFilterResourceRequirements_ and after that remove it in _MiniportStartDevice_. Will this action coerce the NDIS to actually include that resource in the _PNDIS_MINIPORT_INIT_PARAMETERS->AllocatedResources_ parameter (which is passed to the _MiniportInitializeEx_ handler)?
- In case I want to add more MSI sources to the list, will that actually force the OS to provide them or it's just a suggestion that has the same effect as setting the _MessageNumberLimit_ registry key to the desired number?
- The notion of AlternativeLists - do I know what list will actually be chosen? Or do I need to explicitly add/modify/change the resources that I desire in all the AlternativeLists?

jtippet, 2018-03-27:
AllocatedResources reflects what you actually get from the NT arbiter.  It may include resources that you had added.

You cannot merely add an interrupt resource and actually expect that interrupt to work.  Adding a resource to the list means you are taking responsibility for implementing that resource too.  So you would be expected to somehow make that interrupt vector exist.   This is why the docs discourage NIC driver developers from adding resources -- it might at first appear to be a tempting solution to get more interrupt vectors, but all you're doing is cooking the books.  You're not actually getting more hardware resources than the kernel already offered you.

You won't know in advance which AlternativeList will be chosen.  The kernel will attempt to allocate resources from the first (and thus, highest-preference) list.  It will only fall back to the second list if there's a scarcity of resources.  So, NIC drivers typically put the optimistic configuration (e.g., one MSI-X vector per RSS CPU) into the first list, and put a super-simple fallback (e.g., a single interrupt) into the second list.

I-gor-C, 2018-03-27:
Thanks @jtippet , now it's more clear.
But I still miss something in the below part:

>You cannot merely add an interrupt resource and actually expect that interrupt to work. Adding a resource to the list means you are taking responsibility for implementing that resource too. So you would be expected to somehow make that interrupt vector exist.

This is a bit unclear to me. If I know that my HW includes the number of MSI sources that I request, isn't it enough? The existance of interrupt vector is the responsibility of the OS and as long as the HW is compliant and the driver registers the required callbacks, won't the OS create the interrupt vector?

jtippet, 2018-03-27:
The resource assignment process is a negotiation across multiple components.  Even if you know that your PCIe card has a MSI-X table with exactly 32 vectors, other components might not be able to deliver 32 vectors.  E.g., the interrupt controller might be broken, and need to be run in line-based mode.  Or the processor might not have enough IDT entries for the affinity mask you asked for.  Or some exotic PCI bridge is doing something weird between your NIC and the CPU.

Thus, the resource list is implemented as an IRP that goes down and up the device stack, so all interested drivers have a chance to participate in the negotiation.  Device drivers for NT must use NT's resource assignment abstractions (which are admittedly *very* abstract) so that things continue working even in the weird corner cases.

I-gor-C, 2018-03-27:
Got it. One last clarification - if the OS doesn't provide MSIs number according to the MessageNumberLimit registry setting, would it be more attentive to additional interrupts request via 
MiniportFilterResourceRequirements handler?

jtippet, 2018-03-28:
No.  Adding a resource to the resource list doesn't mean you _get_ the resource.  It means you're promising to _provide_ that resource.  In a typical PCIe situation, only the bus itself can actually provide interrupts.

I-gor-C, 2018-03-28:
@jtippet , thanks a lot for the info.

closed 2018-03-28 by I-gor-C

duncanmacmichael, 2018-03-28:
Thanks, @jtippet . @I-gor-C , thanks for the question.