MicrosoftDocs/windows-driver-docs-ddi #36: New behavior (no error) on Windows 10 build 17133 for registry key objects (buffer too small)

kITerE, 2018-03-29:
Source code:
Status = ObQueryNameString(pObject, pObjInfo, uBufSize, &uResLen);

Debugger:
1: kd> dv
        pObject = 0x86a00420
         Status = 0n0
       uBufSize = 0x10
       pObjInfo = 0x8c2c4d30
        uResLen = 0x9e
1: kd> dx pObjInfo
pObjInfo         : 0x8c2c4d30 [Type: _OBJECT_NAME_INFORMATION *]
    [+0x000] Name             : "\RE" [Type: _UNICODE_STRING]
1: kd> !object 0x86a00420
Object: 86a00420  Type: (869169d0) Key
    ObjectHeader: 86a00408 (new version)
    HandleCount: 1 PointerCount: 33
    Directory Object: 00000000  Name: \REGISTRY\MACHINE\SYSTEM\CONTROLSET001\CONTROL\SESSION MANAGER\ENVIRONMENT

That is: STATUS_SUCCESS is returned, but the name is not full and ReturnLength > Length

Windows 10 Kernel build 16299 return expected error - STATUS_INFO_LENGTH_MISMATCH:
0: kd> dv
        pObject = 0xffffce8b`b4a66c40
         Status = 0xc0000004
       uBufSize = 0x10
       pObjInfo = 0xffffce8b`b4d492e0
        uResLen = 0xa6

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 26988d55-c635-d329-f53b-c9a794b80064
* Version Independent ID: ccd1e35e-61f5-3299-3bba-4e628dc43428
* Content: [ObQueryNameString function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-obquerynamestring)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-obquerynamestring.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-obquerynamestring.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

AndrewHarryKim, 2018-04-26:
Hello @kITerE,

Thanks for reaching out to the Windows Driver Developer Documentation team. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.
You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.
• MSDN Windows Hardware WDK and Driver Development forum
• NTDEV forum at OSR Online
Again, thanks for reaching out. Best of luck with this issue.

madcoder3000, 2018-06-10:
@AndrewHarryKim, @kITerE pointed out how the function behavior no longer matches the documentation. It no longer returns STATUS_INFO_LENGTH_MISMATCH which it is supposed to. Why would this not be a documentation issue? The documentation needs to be updated to reflect this change in behavior.

aahill, 2018-06-16:
Hi @kITerE, @madcoder3000, @EliotSeattle 

I've reached out to the development team for comment. Once they respond back I'll update the thread.

Thanks!

aahill, 2018-06-25:
Hi all,

This looks to be a bug - the expected behavior should be STATUS_INFO_LENGTH_MISMATCH as you say. I've passed this along to the development team, so hopefully this will get fixed soon. 

Thanks for finding this!

closed 2018-06-25 by aahill