MicrosoftDocs/windows-driver-docs-ddi #39: damaged link

stonedreamforest, 2018-04-03:
Sections and Views. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3e60ea10-5a11-6331-a47b-9472e444fbcb
* Version Independent ID: 3d19e81c-bdaf-db90-8f39-9b364f3e7404
* Content: [ZwCreateSection function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-zwcreatesection)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-zwcreatesection.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-zwcreatesection.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

spaceisfun, 2018-04-13:
Looks like it is supposed to be linking to https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/section-objects-and-views

tedhudek, 2018-05-18:
Fixed, should be updated within the hour.  Thanks for the find!

closed 2018-05-18 by tedhudek