MicrosoftDocs/windows-driver-docs-ddi #43: IO_IGNORE_SHARE_ACCESS_CHECK behavior

snowkoan, 2018-04-09:
Hi,

The docs are missing an important piece of behavior when IO_IGNORE_SHARE_ACCESS_CHECK is specified.
Yes, it allows the current open to succeed in many cases.
It ALSO means that the file system does not keep track of the current open's access or share access. So, a subsequent open can still succeed even if it normally wouldn't. It's confusing to explain in a couple of sentences. I'll post the same to NTFSD and include a link to my blog, with a much longer explanation.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a61ac7bc-452e-b273-3253-da8e4a277b4f
* Version Independent ID: 7af41083-d22e-7cbf-26dd-9c645baad104
* Content: [FltCreateFileEx2 function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltcreatefileex2)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltcreatefileex2.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltcreatefileex2.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-02:
@aahill I missed this one when I reassigned the issues that Andrew had before he left the team. Can you please pick this up? So sorry, @snowkoan

aahill, 2018-07-03:
Hi @snowkoan, @EliotSeattle, thanks for the notification. I've put this information into the topic and will have it live shortly. Thanks for this pull request, and for documenting the process - we appreciate it!

closed 2018-07-03 by aahill