MicrosoftDocs/windows-driver-docs-ddi #46: Broken link to EVT_WDF_TIMER callback

reuvenab, 2018-04-16:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdftimer/nc-wdftimer-evt_wdf_timer

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dfb50c08-425e-3cf7-7723-0ea1651e0e83
* Version Independent ID: 8a06b0e1-faf0-254a-4a09-da3318a8be9d
* Content: [WdfTimerCreate function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdftimer/nf-wdftimer-wdftimercreate)
* Content Source: [wdk-ddi-src/content/wdftimer/nf-wdftimer-wdftimercreate.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdftimer/nf-wdftimer-wdftimercreate.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-04-16:
Thanks, @reuvenab.
Andrew, this is in the generated TOC entry for EVT_WDF_TIMER. Can you take a look and see what the issue is and that this is not something that affects other entries?  Thanks!

aahill, 2018-06-11:
Hi @reuvenab I'm checking back in on this. It looks like the link should be working now; I've tried both it and a few other topics. If you're still having this issues, feel free to reopen this bug and we'll look into it further. Thanks!

closed 2018-06-11 by aahill