MicrosoftDocs/windows-driver-docs-ddi #51: PsLookupProcessByProcessId: STATUS_INVALID_CID instead of STATUS_INVALID_PARAMETER (Vista and later)

kITerE, 2018-04-25:
On Windows Vista (and later) PsLookupProcessByProcessId returns STATUS_INVALID_CID instead of STATUS_INVALID_PARAMETER

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 96d2e078-66e6-fa76-3415-abf38185f196
* Version Independent ID: 54189ddc-5e5d-3615-993a-3f8f62d23899
* Content: [PsLookupProcessByProcessId function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-pslookupprocessbyprocessid)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-pslookupprocessbyprocessid.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-pslookupprocessbyprocessid.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

AndrewHarryKim, 2018-04-26:
Hello @kITerE, you are correct. It appears the status code was change in vista. The change should be reflected in the next 24 hours.

aahill, 2018-06-11:
Hi @kITerE, Just checking in on this. Looks like the topic has been updated; feel free to reach out with any other issues you come across. Thanks!

closed 2018-06-11 by aahill