MicrosoftDocs/windows-driver-docs-ddi #52: Missing Filter Manager user-mode API header

CAllredMSFT, 2018-04-25:
The documentation for the Filter Manager user-mode library (fltuser.h) is missing.  The APIs themselves are still documented on msdn.microsoft.com (see e.g. FilterLoad at https://msdn.microsoft.com/en-us/library/windows/hardware/ff541504(v=vs.85).aspx), but the new docs.microsoft.com system doesn't have that documentation.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c1a5168c-5f2d-95bc-0546-b5920da1746a
* Version Independent ID: 010ff4bf-422c-e3ef-81c5-6775830e944c
* Content: [Installable file system](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/_ifsk/)
* Content Source: [wdk-ddi-src/content/_ifsk/index.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/_ifsk/index.md)
* Service: **unspecified**
* GitHub Login: @cdwilson
* Microsoft Alias: **windowsdriverdev**

CAllredMSFT, 2018-04-27:
In the meantime MSDN is broken.

If I go to a page for any of the APIs in _fltuser.h_, e.g. [FilterAttach](https://msdn.microsoft.com/en-us/library/windows/hardware/ff540442(v=vs.85).aspx), click on "Minifilter User-Mode Application Functions" near the top of the page, and select the "Minifilter User-Mode Application Functions" from the drop-down, I should be taken to a list of all the user-mode APIs for Filter Manager.  Instead I'm redirected to the [top-level WDK Reference page](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/index).

The content I want to see is (apparently) impossible to reach.

AndrewHarryKim, 2018-04-27:
Apologies for the inconvenience; getting all of the SDK content to docs.microsoft.com is currently a core priority. The navigation issue is one that will go away once these pages are properly migrated to Docs.microsoft.com

If you are trying to access the API content specifically, navigating with the TOC should work fine for now. If you are looking specifically for the conceptual topic on **Minifilter User-Mode Application Functions**, this should still be available in an offline format. 

Your best PoC would be the writer for File Systems for User Mode headers.