MicrosoftDocs/windows-driver-docs-ddi #58: Incorrect information in Remarks

peterwie, 2018-05-09:
This says the PF driver registers the get location callback by setting the WriteVfConfigBlock member of the device interface standard structure.  It should say the GetDeviceLocation member.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ec2082b0-66b5-3097-2acf-bcfc4f067c17
* Version Independent ID: ffd03774-7dfe-4205-4751-f635c222f5e5
* Content: [SRIOV_GET_DEVICE_LOCATION](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/pcivirt/nc-pcivirt-sriov_get_device_location)
* Content Source: [wdk-ddi-src/content/pcivirt/nc-pcivirt-sriov_get_device_location.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/pcivirt/nc-pcivirt-sriov_get_device_location.md)
* Service: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-05-10:
@PageWriter-MSFT 
One quick fix.

PageWriter-MSFT, 2018-05-25:
Remarks is now fixed. Thanks for catching that.

closed 2018-05-25 by PageWriter-MSFT