MicrosoftDocs/windows-driver-docs-ddi #59: Wrong code snippet

peterwie, 2018-05-09:
The code snippet here is for IOCTL_SRIOV_NOTIFICATION and doesn't mention IOCTL_SRIOV_EVENT_COMPLETE

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 54362424-9561-193a-55ab-d02d7a05e015
* Version Independent ID: 36848958-a7bc-8c80-c6fc-55ffbd66646f
* Content: [IOCTL_SRIOV_EVENT_COMPLETE](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/pcivirt/ni-pcivirt-ioctl_sriov_event_complete)
* Content Source: [wdk-ddi-src/content/pcivirt/ni-pcivirt-ioctl_sriov_event_complete.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/pcivirt/ni-pcivirt-ioctl_sriov_event_complete.md)
* Service: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-05-10:
Thanks, @peterwie 
@PageWriter-MSFT Can you take a look and clean this one up? 
Thanks so much, 
Eliot

PageWriter-MSFT, 2018-05-29:
Thanks Peter, I have removed the snippet for now.

closed 2018-05-29 by PageWriter-MSFT