MicrosoftDocs/windows-driver-docs-ddi #61: FLT_PREOP_DISALLOW_FSFILTER_IO

21street, 2018-05-14:
The subject value is not described. As I know it was first introduced in the WDK 15063 (RS2). What does it mean? Why was it added?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 72f7a308-be48-63b5-4602-c1bb7a7dacdb
* Version Independent ID: b66fa659-5880-2c2e-73b8-eb9a7cf95174
* Content: [PFLT_PRE_OPERATION_CALLBACK](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nc-fltkernel-pflt_pre_operation_callback)
* Content Source: [wdk-ddi-src/content/fltkernel/nc-fltkernel-pflt_pre_operation_callback.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nc-fltkernel-pflt_pre_operation_callback.md)
* Service: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

aahill, 2018-07-11:
Hi @21street, I realize my last answer may not have exactly answered your question. I plan on moving the following text to the top of the page to replace the (currently generic) one. Let me know if the following helps:

*A minifilter driver's pre-operation callback routine processes one or more types of I/O operations. This callback routine is similar to a dispatch routine in the legacy filter model.* 

*A minifilter driver registers a pre-operation callback routine for a particular type of I/O operation by storing the callback routine's entry point in the OperationRegistration array of the FLT_REGISTRATION structure. The minifilter driver passes this structure as a parameter to FltRegisterFilter in its DriverEntry routine. A minifilter driver can register a pre-operation callback routine for a given type of I/O operation without registering a post-operation callback (PFLT_POST_OPERATION_CALLBACK) routine and vice versa.*

duncanmacmichael, 2018-10-12:
Hi @21street, I've added a description for FLT_PREOP_DISALLOW_FSFILTER_IO and the changes should be live later today. Thanks again for pointing this out.

closed 2018-10-12 by duncanmacmichael