MicrosoftDocs/windows-driver-docs-ddi #62: FLT_POSTOP_DISALLOW_FSFILTER_IO

21street, 2018-05-14:
The subject value is not described. As I know it was first introduced in the WDK 16299 (RS3). What does it mean? Why was it added?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7c54609f-5bac-41f9-cc9b-65f40e4900dc
* Version Independent ID: fbf86498-496a-fed8-76f0-9a62457db534
* Content: [PFLT_POST_OPERATION_CALLBACK](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nc-fltkernel-pflt_post_operation_callback)
* Content Source: [wdk-ddi-src/content/fltkernel/nc-fltkernel-pflt_post_operation_callback.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nc-fltkernel-pflt_post_operation_callback.md)
* Service: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

aahill, 2018-07-11:
Hi @21street, thanks for reaching out - apologies on the delay in getting to this! Would you let me know what type of information you'd like to see on this page specifically? I notice there's a lot of content in the remarks section (some of which should definitely replace the current generic description). I plan on moving this section of text to the top of the page - let me know what additions you'd like to see:

*A minifilter driver's pre-operation callback routine processes one or more types of I/O operations. This callback routine is similar to a dispatch routine in the legacy filter model.* 

*A minifilter driver registers a pre-operation callback routine for a particular type of I/O operation by storing the callback routine's entry point in the OperationRegistration array of the FLT_REGISTRATION structure. The minifilter driver passes this structure as a parameter to FltRegisterFilter in its DriverEntry routine. A minifilter driver can register a pre-operation callback routine for a given type of I/O operation without registering a post-operation callback (PFLT_POST_OPERATION_CALLBACK) routine and vice versa.*

Thanks!

aahill, 2018-07-11:
Hi @21street, From what I understand, this  callback represents the post-operation state of the I/O process represented by the IO_STATUS_BLOCK, and performs completion processing. Would adding this information, along with the content in #61 help?

Thanks!

duncanmacmichael, 2018-10-12:
Hi @21street, I've added a description for FLT_POSTOP_DISALLOW_FSFILTER_IO and the changes should be live later today. Thanks again for pointing this out.

closed 2018-10-12 by duncanmacmichael