MicrosoftDocs/windows-driver-docs-ddi #65: PreQueryOpen and PostQueryOpen fields in FS_FILTER_CALLBACKS

21street, 2018-05-18:
There are no subject fields in the FS_FILTER_CALLBACKS structure.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f3aed758-d5dc-0128-62fb-2bc69c635540
* Version Independent ID: 8351a690-5a4c-8c30-11f6-eb0a62ed3ded
* Content: [FsRtlRegisterFileSystemFilterCallbacks function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-fsrtlregisterfilesystemfiltercallbacks#parameters)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-fsrtlregisterfilesystemfiltercallbacks.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-fsrtlregisterfilesystemfiltercallbacks.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

aahill, 2018-06-11:
Hi @21street, thanks for reaching out. I've added the missing fields into the topic. Could you give me a little bit of context (a sentence or two) about the operation, so that I can add a description to it?

Thanks!

duncanmacmichael, 2018-10-12:
Hi @21street , sorry for the delay in getting this fixed. I've added a description for the PreQueryOpen and PostQuery fields, and those changes should be live later today. Thanks again for your interest in the docs!

closed 2018-10-12 by duncanmacmichael