MicrosoftDocs/windows-driver-docs-ddi #70: PageProtection list is not complete

21street, 2018-05-18:
There can be some other bit flags, for example, PAGE_EXECUTE_READWRITE.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f3aed758-d5dc-0128-62fb-2bc69c635540
* Version Independent ID: 8351a690-5a4c-8c30-11f6-eb0a62ed3ded
* Content: [FsRtlRegisterFileSystemFilterCallbacks function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-fsrtlregisterfilesystemfiltercallbacks#parameters)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-fsrtlregisterfilesystemfiltercallbacks.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-fsrtlregisterfilesystemfiltercallbacks.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

aahill, 2018-06-11:
Thanks @21street, I can help add the flag you mention to the topic. I'm having trouble tracking PageProtection down - are you aware of any other bit flags that should be documented here?

aahill, 2018-06-13:
Quick update, I was able to locate the PageProtection information, and I've updated the topic. Should be live soon. Thanks!

closed 2018-06-13 by aahill