MicrosoftDocs/windows-driver-docs-ddi #74: FLT_PARAMETERS.QueryOpen structure is not described

21street, 2018-05-18:
There is no QueryOpen structure in the FLT_PARAMETERS one

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 90b8d975-b4f8-c43f-f91f-fad1bb5a5cee
* Version Independent ID: 8efd9348-dcb5-d5fc-5e1e-6b13ab2e5321
* Content: [_FLT_PARAMETERS](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/ns-fltkernel-_flt_parameters)
* Content Source: [wdk-ddi-src/content/fltkernel/ns-fltkernel-_flt_parameters.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/ns-fltkernel-_flt_parameters.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

rodwiddowson, 2018-06-12:
The major code is IRP_MJ_QUERY_OPEN

It would be extremely useful to have a semi-definitive list of the `FileInformationClass` values that can be sent.  My assumption would be  `FileStatInformation ` and `FileStatLxInformation` (>= RS4)

aahill, 2018-06-12:
Hi @21street, @rodwiddowson 

I'd be happy to investigate this, and create the documentation. For OpenQuery, I see the union component is:

```
struct {
        PIRP Irp;
        PVOID FileInformation;
        PULONG Length;
        FILE_INFORMATION_CLASS FileInformationClass;
    } QueryOpen;
```
I'm new to this area (taking it on for Andrew), so I'm uncertain what I should include for a description of these parameters. Would either of you be able to suggest some? If not, I can reach out to the feature team - just want to make sure they're descriptive :) 

Thanks!

rodwiddowson, 2018-06-12:
@aahill I think you will need to reach out to the devs. This is a completely new path through the io system and so is treading new ground.   Aggravatingly everything else I can say is under NDA (with Microsoft) so you need to keep this in house.

rodwiddowson, 2018-07-02:
I should add (since I'm in the area)

- It is not unheard of to see an FO with a related FO in it down this code path. (I particularly love the null file name thing too)
- Documentation needs to cover any need to call the oplock package.

Same is true for #65 and #68 (which are really all the same case as this)

duncanmacmichael, 2018-10-12:
Hi @21street and @rodwiddowson, sorry for the delay in getting this answered for you. Though my usual area is network driver docs, I've helped to fill in a few missing descriptions for some of the IFSK docs. I worked with the devs to document everything on [FsRtlRegisterFileSystemFilterCallbacks](https://docs.microsoft.com/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-fsrtlregisterfilesystemfiltercallbacks), so hopefully that should answer your questions about the PreQueryOpen and PostQueryOpen operations. And @rodwiddowson, you were nearly correct for the FileInformationClassValues. :) Those two plus FileCaseSensitiveInformation are the only three possible values.

Thanks to you both for your interest in the docs! I'll go ahead and close this issue now.

closed 2018-10-12 by duncanmacmichael