MicrosoftDocs/windows-driver-docs-ddi #76: No feedback for this topic

mfstanton, 2018-05-18:
The sections begins with "In this section..." and all I get is a bunch of hyperlinks that point to another page. Do you realize how painful it makes this documentation to follow? I should be able to follow it like a book by simply scrolling down through the pages.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 647e6c5b-caa6-6f3b-9d00-370ce1ed8add
* Version Independent ID: db78d666-4a92-eb4d-7886-40f543cce83a
* Content: [wdf.h header](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdf/index)
* Content Source: [wdk-ddi-src/content/wdf/index.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdf/index.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @cdwilson
* Microsoft Alias: **windowsdriverdev**

cdwilson, 2018-05-20:
@mfstanton why do I keep getting tagged in these windows-driver-docs-ddi issues?

mfstanton, 2018-05-20:
I don’t know. I suspect someone at Microsoft has you in for it. All I did was hit the windows DDI documentation Feedback link and off they go. I suspect you were the target of the email address they spawned from the documentation page.. So sorry for disturbing you. I will see if I can get the attention of someone at Microsoft.  They recently moved their whole repository over to Github.

tedhudek, 2018-05-21:
Thanks @cdwilson for letting us know. It turns out there was a bug in our internal tooling that caused this to happen. I'm hopeful you won't see any more of these in the meantime!

@mfstanton can you give me some more specifics on the layout you'd like to see? We recently switched to a header-based system for the driver API docs, and we're eager for feedback. I'm not sure I understand what you'd like to see.

closed 2018-06-22 by tedhudek