MicrosoftDocs/windows-driver-docs-ddi #77: what should I do with DeviceDirectoryHandle?

cashingad, 2018-05-22:
Open file, create file etc. requires path and file name.
How can I get the path from the handle? 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a37359e8-8ea8-7240-7302-e3a1a6225060
* Version Independent ID: 0822ae8c-973d-5051-ea91-e98bc4a0ee12
* Content: [IoGetDeviceDirectory function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-iogetdevicedirectory)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-iogetdevicedirectory.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-iogetdevicedirectory.md)
* Service: **unspecified**
* Product: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

msshapira, 2018-05-23:
this seems to be different that the new WDF API ( [WdfDeviceRetrieveDeviceDirectoryString  ](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdfdevice/nf-wdfdevice-wdfdeviceretrievedevicedirectorystring) ) where you get a string and can Enumerate the directory or access files with regular Zw API. Here you get a directory handle which doesn't seem to have much use. 

Even the user mode API are quite limited for a directory handle (see https://msdn.microsoft.com/en-us/library/windows/desktop/aa365258(v=vs.85).aspx )

cashingad, 2018-05-24:
I think we found a solution, on the object attribute in ZwCreateFile/ZwOpenFile etc. we configure the RootDirectory:
#define InitializeObjectAttributes( p, n, a, r, s ) { \
    (p)->Length = sizeof( OBJECT_ATTRIBUTES );          \
    **(p)->RootDirectory = r;                             \**
    (p)->Attributes = a;                                \
    (p)->ObjectName = n;                                \
    (p)->SecurityDescriptor = s;                        \
    (p)->SecurityQualityOfService = NULL;               \
    }

PageWriter-MSFT, 2018-05-25:
Hello, 

I checked with the product team, your approach is correct. This WDM API is called by kernel drivers. Drivers typically use ZwOpenFile and ZwCreateFile to access/create files.  One of the parameters for those functions is a OBJECT_ATTRIBUTES. In the object attributes is object name and a root directory, as you have shown.  If the root directory is NULL, then the object name must be a fully qualified path.  However, if you provide a handle for the root directory, then the object name has to be relative to the object (in the case of files, the directory), that the handle represents.  So the intent with IoGetDeviceDirectory (and IoGetDriverDirectory) is that you take the out HANDLE and use it as a root directory in the object attributes you are passing to a ZwOpenFile or ZwCreateFile. 

Let us know if you have any more questions,

PageWriter

closed 2018-05-25 by PageWriter-MSFT