MicrosoftDocs/windows-driver-docs-ddi #80: Bad link in initial description sentence.

BenWestbrook, 2018-05-23:
"The ZwMapViewOfSection routine maps a view of a section into the virtual address space of a subject process."

The link on the word "view" in the above sentence goes to a completely unrelated topic.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c3ee14b8-7fe3-d780-6166-6587076199ef
* Version Independent ID: e9beb1a4-8c89-f40f-ff5d-7e0d4b37e002
* Content: [ZwMapViewOfSection function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-zwmapviewofsection)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-zwmapviewofsection.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-zwmapviewofsection.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

PageWriter-MSFT, 2018-05-23:
@BenWestbrook, thanks for catching that. It's fixed. Should be live soon.

closed 2018-05-23 by PageWriter-MSFT

GabeSJones, 2018-08-15:
Apparently it was fixed by removing the word "view" entirely:

"The ZwMapViewOfSection routine maps a of a section"

EliotSeattle, 2018-10-30:
Ooops. Thanks, @GabeSJones. The word is back, minus the link.