MicrosoftDocs/windows-driver-docs-ddi #81: missing critical information

BenWestbrook, 2018-05-23:
TBD

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 75aca267-3561-e63a-4737-e1844c635d49
* Version Independent ID: 94de8e18-cf27-3b40-5e88-f53e1bff9827
* Content: [InitializeObjectAttributes macro](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wudfwdm/nf-wudfwdm-initializeobjectattributes)
* Content Source: [wdk-ddi-src/content/wudfwdm/nf-wudfwdm-initializeobjectattributes.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wudfwdm/nf-wudfwdm-initializeobjectattributes.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-05-23:
Thanks, @BenWestbrook 
@tedhudek please rid us of these TBDs! Thanks, sir.

tedhudek, 2018-06-19:
Fixed.  Will be live within 24 hrs

closed 2018-06-19 by tedhudek