MicrosoftDocs/windows-driver-docs-ddi #82: Opportunity for helpful link missed

BenWestbrook, 2018-05-24:
"...returns the current IRQL"

In the above sentence, if IRQL were a link to the page describing IRQL values, this page would be much more useful.

As it is now, the reader is left to "go fish". 8-)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 41f9cc85-d0b3-88db-cf7f-ee68443a0ddb
* Version Independent ID: 2fc5b77a-0ef9-8ade-62bd-e13391f8f83e
* Content: [KeGetCurrentIrql function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-kegetcurrentirql)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-kegetcurrentirql.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-kegetcurrentirql.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-05-24:
Thanks, @BenWestbrook 
@PageWriter-MSFT I believe this function is in PCI, but for some reason is not rendering in the TOC. Can you fix that and look at the link, please?

PageWriter-MSFT, 2018-05-25:
@BenWestbrook 

Thanks for the feedback. I have added a link to [Managing Hardware Priorities](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/managing-hardware-priorities), which describes the IRQLs at which most drivers are called. The update should be live soon.

Let us know if you have other suggestions!

PageWriter

closed 2018-05-25 by PageWriter-MSFT