MicrosoftDocs/windows-driver-docs-ddi #90: InstanceQueryTeardownCallback description

21street, 2018-05-25:
There's no comment "If a NULL is specified for this routine, then instances can never be manually detached".

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 020480f3-b724-f282-cce8-43d7cde038d2
* Version Independent ID: 84eeb4e3-4d88-862c-276e-7496c0f35f9b
* Content: [_FLT_REGISTRATION](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/ns-fltkernel-_flt_registration)
* Content Source: [wdk-ddi-src/content/fltkernel/ns-fltkernel-_flt_registration.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/ns-fltkernel-_flt_registration.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

aahill, 2018-06-11:
Hi @21street, thanks for reaching out. I've added your comment to the "remarks" Section on the page. It should be live shortly. Thanks for pointing this out!

closed 2018-06-11 by aahill

aahill, 2018-06-11:
Quick update, I mean I've added the comment to the field description