MicrosoftDocs/windows-driver-docs-ddi #91: FwpsInjectMacSendAsync and FwpsInjectMacReceiveAsync are the same

BOT-Man-JL, 2018-05-26:
Docs of `FwpsInjectMacSendAsync` look like a copy of `FwpsInjectMacReceiveAsync` with only modifying the `Syntax` section... Are they actually work in the same way? But according to my investigation and WFPSampler, description _The NET_BUFFER_LIST structure must begin with an IP header_ of parameter _netBufferLists_ should be _begin with an ethernet header_... Thanks!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2bd93606-22dc-f6d6-cef5-f4324b7dcc63
* Version Independent ID: 3aeb627c-7a54-970a-4c80-943d068876e3
* Content: [FwpsInjectMacSendAsync0 function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fwpsk/nf-fwpsk-fwpsinjectmacsendasync0#feedback)
* Content Source: [wdk-ddi-src/content/fwpsk/nf-fwpsk-fwpsinjectmacsendasync0.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fwpsk/nf-fwpsk-fwpsinjectmacsendasync0.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

duncanmacmichael, 2018-05-31:
Hi @BOT-Man-JL , I've gotten confirmation on this for you that the packet should begin with a MAC header when doing outbound injection at this layer. I'll update the doc. Thanks again for your interest in improving the docs!

closed 2018-05-31 by duncanmacmichael

BOT-Man-JL, 2018-06-01:
Hi @duncanmacmichael , thank you for your reply.

According to my investigation and the samples, both inbound and outbound layer2 injection are beginning with MAC header...

And the docs of `FwpsInjectMacSendAsync` look like copied from `FwpsInjectMacReceiveAsync `... Would you like to check more details and update both? Thanks!

duncanmacmichael, 2018-06-01:
Hi @BOT-Man-JL, that's partially true. For inbound injection, the NBL should start with a MAC header in the Native case, or an IP header in the Ethernet case. This matches the code in the WFP sampler, which retreats the NBL only if Native. Like so:

```
/// If NATIVE, initial offset is at the MAC Header ...
   if(pClassifyValues->layerId != FWPS_LAYER_INBOUND_MAC_FRAME_NATIVE &&
      FWPS_IS_L2_METADATA_FIELD_PRESENT(pMetadata,
                                        FWPS_L2_METADATA_FIELD_ETHERNET_MAC_HEADER_SIZE))
      bytesRetreated = pMetadata->ethernetMacHeaderSize;

   if(bytesRetreated)
   {
      /// ... otherwise the offset is at the IP Header, so retreat the size of the MAC Header ...
      status = NdisRetreatNetBufferDataStart(NET_BUFFER_LIST_FIRST_NB((NET_BUFFER_LIST*)pCompletionData->pClassifyData->pPacket),
                                             bytesRetreated,
                                             0,
                                             0);

. . . (more sample code, including advancing the original NBL later if it were retreated)
```

I'll update the FwpsInjectMacReceiveAsync function to clarify this. And yes, the injection functions at the MAC layer are identical otherwise, which is probably why the original author duplicated most of the text. Their parameters are identical in the headers and their usage is identical in the samples, other than having to worry about retreating the NBL for inbound injection at the Ethernet layer.

To re-clarify from the [Data Offset Positions](https://docs.microsoft.com/windows-hardware/drivers/network/data-offset-positions) page:

Native – offset is at the MAC Header
Ethernet – offset is at the IP Header
Outbound:
Native – offset is at the MAC Header
Ethernet – offset is at the MAC Header

So for the four scenarios where you're performing injection at L2, three of them require the NBL to have the header at the beginning of the MAC header and one requires it to have the header at the beginning of the IP header.

Hope this helps.

duncanmacmichael, 2018-06-04:
Hi @BOT-Man-JL , no problem! Happy to help make the docs clearer. That's correct according to the sample code. 😊