MicrosoftDocs/windows-driver-docs-ddi #92: Clarification needed

BenWestbrook, 2018-05-27:
From the item description, this routine allows the driver to setup driver tracing and any related resources. The puzzler for me has to do with where do I park the pointers to these resources so that I can free them in CleanupTracing. Normally I would put resources in the appropriate extension such as DriverExtension, SrbExtension, etc. However, this routine is called by StorPort prior to the return from StorPortInitialize which is called from the driver in DriverEntry...our DriverExtension pointer has not yet been given to us. This happens at the next step FindAdapter. If the struct passed as the first pointer to DriverEntry (and subsequently passed to StorPortInitialize [and subsequently passed to InitializeTracing])  contains a pointer location for us to save our tracing allocated resources, then this page should document that. Presently, if tracing is to be used either a global must be used which is incompatible with multiple instance of the driver, or the driver will leak memory.

I am sure I am missing something simple and/or obvious, but, some clarification would help. Surely I am not the only one puzzled by this.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: fa837d0c-345b-8324-a85d-57509a71542c
* Version Independent ID: 3b866097-878a-f845-d892-2c4b60c0c452
* Content: [HW_INITIALIZE_TRACING](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/storport/nc-storport-hw_initialize_tracing)
* Content Source: [wdk-ddi-src/content/storport/nc-storport-hw_initialize_tracing.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/storport/nc-storport-hw_initialize_tracing.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

aahill, 2018-06-13:
Hi @BenWestbrook, I received the following from the development team:

"we’re not expecting any resources to be needed to be able to initialize tracing. If miniport needs to allocate resource, then it’ll have to be a global as the callback is applicable to all instances of the driver."

Does this help answer your question?

BenWestbrook, 2018-06-13:
This does answer the question but not the problem. Clearly, the API would have provided access to the device extension if the development team anticipated the need.

 

However, there is a workaround if you do not need tracing in DriverEntry and FindAdapter. Shortly after the return from FindAdapter the function TracingEnableDisable is called, with the deviceExtension. Tracing can be enabled and allocations made at this point, and then cleaned up when called to disable tracing or driver unload.

 

Thank you for the follow up.

aahill, 2018-06-14:
No problem @BenWestbrook, glad I could provide some information. Thanks for posting your workaround, it'll help folks who look for similar info!

closed 2018-06-14 by aahill