MicrosoftDocs/windows-driver-docs-ddi #94: EG_CREATE_KEY_INFORMATION[_V1] Absolute path character

kITerE, 2018-05-28:
> If the path is absolute, this structure contains a fully qualified path that starts with the **""** character. 
> If the path is relative, the path starts with a character other than **""**

Symbol is not displayed, character "\" escaping is required

REG_CREATE_KEY_INFORMATION (https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/ns-wdm-_reg_create_key_information) - same problem

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 577c6e3c-5587-723c-d8f0-0b409da989b4
* Version Independent ID: af591655-5cf7-51b0-5156-d1357cd76846
* Content: [_REG_CREATE_KEY_INFORMATION_V1](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/ns-wdm-_reg_create_key_information_v1)
* Content Source: [wdk-ddi-src/content/wdm/ns-wdm-_reg_create_key_information_v1.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_reg_create_key_information_v1.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

kITerE, 2018-05-28:
Pull request: #95

tedhudek, 2018-05-29:
Assuming this issue preceded your PR, thanks again

closed 2018-05-29 by tedhudek