MicrosoftDocs/windows-driver-docs-ddi #97: Is this IOCTL deprecated?

R030t1, 2018-05-30:
The requirements section on this page notes the IOCTL code is supported above Windows 8, on Windows 2003, and on Windows XP, with a gap for Windows 7, Vista, and Server 2008. However the page for USB_CYCLE_PORT_PARAMS notes that this IOCTL is deprecated. Which is right?

Please do not deprecate this IOCTL - it is very useful.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a43e4c0e-3474-a2af-0407-837a45eb7645
* Version Independent ID: 151b4d70-de45-4923-5099-e7ff50bec31d
* Content: [IOCTL_USB_HUB_CYCLE_PORT](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/usbioctl/ni-usbioctl-ioctl_usb_hub_cycle_port)
* Content Source: [wdk-ddi-src/content/usbioctl/ni-usbioctl-ioctl_usb_hub_cycle_port.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/usbioctl/ni-usbioctl-ioctl_usb_hub_cycle_port.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

R030t1, 2018-06-04:
I have more information for consideration. Per the USB DFU specification<sup>[1]</sup> (section 2, "Overview", steps 2 and 4), hosts must be able to issue a reset to a device. The name ("USB reset") implies it is part of the USB protocol itself, and indeed this state exists and is generated by bringing both D- and D+ low.<sup>[2], [3]</sup>

Reading more about the Windows USB driver stack I have become confused. It seems there is not necessarily any way to generate a state guaranteed to exist by the standard. A thread on OSR Online's NTDEV list<sup>[4]</sup> (potentially authored during the unavailability of IOCTL_USB_HUB_CYCLE_PORT) discusses this issue and some alternatives, but it seems like they are not drop in replacements and do not guarantee reenumeration.

[1]: http://www.usb.org/developers/docs/devclass_docs/DFU_1.1.pdf
[2]: http://microchipdeveloper.com/usb:reset-suspend-resume
[3]: http://www.usbmadesimple.co.uk/ums_3.htm
[4]: https://www.osronline.com/showthread.cfm?link=252223

PageWriter-MSFT, 2018-06-05:
Hello @R030t1,

This IOCTL was deprecated for Windows 7 but revived in Windows 8 with the caveat that only callers with Admin privileges can use it. 

This is mentioned in the Requirements section.
_"Windows version | Supported on Windows 8 and later versions of Windows,  if the caller is running as Administrator."_

The associated structure is marked as deprecated. This is misleading. I have updated the content to remove that note. 

Thanks for finding this and your research!

Page Writer

closed 2018-06-05 by PageWriter-MSFT