MicrosoftDocs/windows-driver-docs-ddi #99: Dead link

peterwie, 2018-05-31:
THe link at the top of the page to GUID_BUS_INTERFACE_STANDARD (https://msdn.microsoft.com/library/windows/hardware/ff546561) goes to a non-existent page.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d8cdc030-6da5-35cd-604a-445a0109c852
* Version Independent ID: fc3feb34-1b08-ab96-0d77-4c42a811857c
* Content: [_BUS_INTERFACE_STANDARD](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/ns-wdm-_bus_interface_standard)
* Content Source: [wdk-ddi-src/content/wdm/ns-wdm-_bus_interface_standard.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_bus_interface_standard.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-06-01:
Fixed!  Thanks Peter

closed 2018-06-01 by tedhudek

peterwie, 2018-06-01:
It now points to a topic page.  I was expecting that it would point to a reference page, preferably one that would tell me which header file to include to get the definition of GUID_BUS_INTERFACE_STANDARD (turns out it's wdmguids.h but I had to grep the headers to find it)

tedhudek, 2018-06-01:
Thanks for the heads-up. I added the header to the target page, will be live by EOD