MicrosoftDocs/windows-driver-docs-ddi #101: Member descriptions reversed

Shenn299, 2018-06-05:
VendorName.Length description and VendorName.Buffer are reversed, isn't it ?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3edb3dd1-3325-c34b-23fb-a409a0c13d7e
* Version Independent ID: a58179dc-1606-e0ec-dacc-6a391707b8d7
* Content: [_VENDOR_ATTR](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/smclib/ns-smclib-_vendor_attr)
* Content Source: [wdk-ddi-src/content/smclib/ns-smclib-_vendor_attr.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/smclib/ns-smclib-_vendor_attr.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-05:
Thanks, @Shenn299 
@aahill can you double check which parts of the two descriptions should be switched?

aahill, 2018-06-05:
HI @Shenn299, Thanks for reaching out. It looks like these two descriptions got reversed at some point - thanks for catching that. I've just fixed them, and they should be visible in a day or so. 

Thanks again, and feel free to reach out with any other problems you come across!

closed 2018-06-05 by aahill

Shenn299, 2018-06-05:
@aahill @EliotSeattle you're welcome ! Thanks for your quick fix.
If I have question about the KMDF PCMCIA Smart Card Driver, could you provide me any support or a contact ?

EliotSeattle, 2018-06-05:
Hi @Shenn299 
We can answer questions you have about the documentation. Unfortunately, we don't have the staff to answer the many questions that arise about people's code. For that, we recommended either the [OSR Ntdev forum](http://www.osronline.com/cf.cfm?PageURL=showlists.cfm?list=NTDEV) or the [Windows Hardware WDK and Driver Development forum ](https://social.msdn.microsoft.com/Forums/windowshardware/en-US/home?forum=wdk).

Shenn299, 2018-06-05:
@EliotSeattle, ok thanks for your reply.