MicrosoftDocs/windows-driver-docs-ddi #102: Directly indicate that output buffer must be set for IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX.

R030t1, 2018-06-06:
My reading of the documentation was that the input buffer was also used as the output buffer. A quick test indicated otherwise, but it may help to amend the documentation.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 210d1acd-67a1-6a3d-adeb-84bf6048c483
* Version Independent ID: 4a65abb1-9f7a-c852-404e-1a76ce9874cc
* Content: [IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/usbioctl/ni-usbioctl-ioctl_usb_get_node_connection_information_ex)
* Content Source: [wdk-ddi-src/content/usbioctl/ni-usbioctl-ioctl_usb_get_node_connection_information_ex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/usbioctl/ni-usbioctl-ioctl_usb_get_node_connection_information_ex.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-06:
Thanks, @R030t1 
@PageWriter-MSFT can you look into this please?

PageWriter-MSFT, 2018-06-06:
Hi @R030t1 

For this IOCTL request, the input buffer is also the output buffer. On input the caller specifies the port index. On output, the structure is filled with information about that port. 

I have updated the topic to indicate that the structures is the Input / Output Buffer.

Also added a snippet that might be useful. The port index is specified by the caller.

```
    HANDLE handle = CreateFile(
                            Path,
                            GENERIC_WRITE | GENERIC_READ,
                            FILE_SHARE_WRITE | FILE_SHARE_READ,
                            NULL,
                            OPEN_EXISTING,
                            NULL,
                            NULL
                        );

    PUSB_NODE_CONNECTION_INFORMATION_EX ConnectionInfo =
                    (PUSB_NODE_CONNECTION_INFORMATION_EX)
                    malloc(sizeof(USB_NODE_CONNECTION_INFORMATION_EX));

    ConnectionInfo->ConnectionIndex = PortIndex;

    ULONG bytes = 0;
    BOOL success = DeviceIoControl(handle,
                              IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX,
                              ConnectionInfo,
                              sizeof(USB_NODE_CONNECTION_INFORMATION_EX),
                              ConnectionInfo,
                              sizeof(USB_NODE_CONNECTION_INFORMATION_EX),
                              &bytes,
                              NULL);

    CloseHandle(handle);

    *pPortSpeed = (UCHAR)ConnectionInfo->Speed;

```

Please let us know if you need more information,

Priyanka

closed 2018-06-06 by PageWriter-MSFT