MicrosoftDocs/windows-driver-docs-ddi #106: missing explicit guidance, user left to guess

BenWestbrook, 2018-06-06:
For a Storport Virtual Miniport (the type of driver that would be using this structure) the value for AdapterInterfaceType will almost always be "internal" (without the quotes). Simply stating that in the description for AdapterInterfaceType seems more useful and straightforward than the current description, at least to me.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: edda33b3-246f-1229-d613-79fcd7619de2
* Version Independent ID: e7199ca7-bd33-9981-0b6b-99ee8024227e
* Content: [_VIRTUAL_HW_INITIALIZATION_DATA](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/storport/ns-storport-_virtual_hw_initialization_data)
* Content Source: [wdk-ddi-src/content/storport/ns-storport-_virtual_hw_initialization_data.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/storport/ns-storport-_virtual_hw_initialization_data.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-06:
Thanks @BenWestbrook 
@aahill One more on this topic. :-)

aahill, 2018-06-06:
@BenWestbrook , thanks for the feedback. I've incorporated this information into the topic and you should be able to see it soon!

closed 2018-06-06 by aahill