MicrosoftDocs/windows-driver-docs-ddi #107: less than direct guidance

BenWestbrook, 2018-06-06:
For the type of driver that would use this structure, MapBuffers should be set to STOR_MAP_ALL_BUFFERS_INCLUDING_READ_WRITE.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: edda33b3-246f-1229-d613-79fcd7619de2
* Version Independent ID: e7199ca7-bd33-9981-0b6b-99ee8024227e
* Content: [_VIRTUAL_HW_INITIALIZATION_DATA](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/storport/ns-storport-_virtual_hw_initialization_data)
* Content Source: [wdk-ddi-src/content/storport/ns-storport-_virtual_hw_initialization_data.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/storport/ns-storport-_virtual_hw_initialization_data.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-06:
Thanks, @BenWestbrook 
@aahill can you please investigate?

aahill, 2018-06-11:
Hi @BenWestbrook, I've added this information to the topic - should be visible soon. I've also added some information about the other mapbuffers, such as STOR_MAP_NON_READ_WRITE_BUFFERS, and STOR_MAP_NO_BUFFERS.

Thanks for bringing this to our attention, feel free to reach out again!

closed 2018-06-11 by aahill