MicrosoftDocs/windows-driver-docs-ddi #108: not certain the description is accurate

BenWestbrook, 2018-06-06:
Do the driver folks believe ReceiveEvent must be set to TRUE for a storport virtual miniport driver?

In my experience setting to FALSE does not change the way the driver interacts with storport - even if set to TRUE storport never calls the ReceiveEvent callback in a storport vitual miniport driver.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: edda33b3-246f-1229-d613-79fcd7619de2
* Version Independent ID: e7199ca7-bd33-9981-0b6b-99ee8024227e
* Content: [_VIRTUAL_HW_INITIALIZATION_DATA](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/storport/ns-storport-_virtual_hw_initialization_data)
* Content Source: [wdk-ddi-src/content/storport/ns-storport-_virtual_hw_initialization_data.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/storport/ns-storport-_virtual_hw_initialization_data.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-06:
Thanks for finding this, @BenWestbrook 
@aahill can you verify and update the content accordingly?

aahill, 2018-06-13:
Hi @BenWestbrook , apparently this field looks to be a holdover from an earlier implementation. Because of this, it's fine to set it to FALSE for storport virtual miniports. I'll update the documentation accordingly.

Thanks for bringing this to our attention!

closed 2018-06-13 by aahill