MicrosoftDocs/windows-driver-docs-ddi #113: TBD for parameters

vijaycognore, 2018-06-08:
Hi, there is no description of the parameters and mentioned as TBD. Check!!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c05f4f5a-e51d-5db9-70cb-39253030ff23
* Version Independent ID: 82a186bd-e1a1-5f0c-6758-f2b2a04f5a55
* Content: [IoCallDriver macro](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-iocalldriver)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-iocalldriver.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-iocalldriver.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-08:
@vijaycognore YIKES! Thanks for catching this. @PageWriter-MSFT can you please get some descriptive text for these?

rig23, 2018-08-10:
Is the return value really void?

vijaycognore, 2018-09-11:
I don't think return value is void. It must return NTSTATUS. This is an important function for device drivers and needs to be corrected.