MicrosoftDocs/windows-driver-docs-ddi #115: Is S_OK intentially recommended for interface methods?

emuzychenko, 2018-06-09:
All IMiniportStreamAudioEngineNode method descriptions recommend to return S_OK as a success indication. Although S_OK equals STATUS_SUCCESS, isn't it better to use STATUS_SUCCESS constant, as usual?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 669af988-2348-c3df-2e2e-dfa417fa9d0a
* Version Independent ID: c187a979-2c81-ebc5-a68c-42db0f0c0992
* Content: [IMiniportStreamAudioEngineNode](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/portcls/nn-portcls-iminiportstreamaudioenginenode)
* Content Source: [wdk-ddi-src/content/portcls/nn-portcls-iminiportstreamaudioenginenode.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/portcls/nn-portcls-iminiportstreamaudioenginenode.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

assarbad, 2018-07-03:
`HRESULT` and `NTSTATUS` are _fully compatible_ and follow the exact same layout.

First I thought: Why would it matter which one is being used?

But I agree that this seems to be an **inconsistency**, given that the methods are being declared as ` STDMETHODIMP_(NTSTATUS)` and `STDMETHOD_(NTSTATUS,<Name>)`. So for consistency's sake this should probably use the `STATUS_SUCCESS` name rather than `S_OK`.

DOMARS, 2020-12-21:
@emuzychenko - Apologies in that I have not been able to gather more information on this. I think the response from @assarbad is reasonable. If this is still an issue, I would post this question to the new audio driver forum that is available in the Windows forums: https://docs.microsoft.com/en-us/answers/products/windows

closed 2020-12-21 by DOMARS