MicrosoftDocs/windows-driver-docs-ddi #116: Type is wrong

hsebs, 2018-06-10:
It is union, not struct.

Line 37007 of wdm.h in wdk 10.0.17134.0

typedef union _OB_PRE_OPERATION_PARAMETERS {
    _Inout_ OB_PRE_CREATE_HANDLE_INFORMATION        CreateHandleInformation;
    _Inout_ OB_PRE_DUPLICATE_HANDLE_INFORMATION     DuplicateHandleInformation;
} OB_PRE_OPERATION_PARAMETERS, *POB_PRE_OPERATION_PARAMETERS;

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9284b14d-d4ec-3d61-b29c-8921906e10c5
* Version Independent ID: fde29a69-8b56-6464-1235-21ed07d3dc79
* Content: [_OB_PRE_OPERATION_PARAMETERS](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/ns-wdm-_ob_pre_operation_parameters)
* Content Source: [wdk-ddi-src/content/wdm/ns-wdm-_ob_pre_operation_parameters.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_ob_pre_operation_parameters.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-10:
Thanks, @hsebs for catching this.
@PageWriter-MSFT, can you please take a look. Title and syntax look wrong, but the description seems correct.

PageWriter-MSFT, 2018-06-11:
@hsebs. It's a bug in our internal build system. I have opened a ticket to get this fixed.

Thanks for catching that!

Page Writer

PageWriter-MSFT, 2018-07-17:
This is bug is now fixed. The union shows up as:

```
typedef union _OB_PRE_OPERATION_PARAMETERS {
  OB_PRE_CREATE_HANDLE_INFORMATION    CreateHandleInformation;
  OB_PRE_DUPLICATE_HANDLE_INFORMATION DuplicateHandleInformation;
} OB_PRE_OPERATION_PARAMETERS, *POB_PRE_OPERATION_PARAMETERS;
```

Thanks for finding that.

Page Writer

closed 2018-07-17 by PageWriter-MSFT