MicrosoftDocs/windows-driver-docs-ddi #117: Params are missing.

csm10495, 2018-06-11:
The params seem to be on the md file: https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/storport/nf-storport-storportreadregisterbufferuchar.md. Though they are out of order there too.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: eeb23dbb-69e7-2f6a-7083-ce76d631d677
* Version Independent ID: ed8483de-12ec-0452-1e79-5b862068bf31
* Content: [StorPortReadRegisterBufferUchar macro](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/storport/nf-storport-storportreadregisterbufferuchar)
* Content Source: [wdk-ddi-src/content/storport/nf-storport-storportreadregisterbufferuchar.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/storport/nf-storport-storportreadregisterbufferuchar.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-11:
Thanks, @csm10495 
@aahill would you be so kind as to look into these TBDs?

aahill, 2018-06-11:
Hi @csm10495, @EliotSeattle 

Thanks for sending this to me. It looks like a number of topics here aren't having their param info showing up. I've gone through and fixed them; they should show up in a day or two. Thanks again for bringing  this to our attention!

closed 2018-06-11 by aahill