MicrosoftDocs/windows-driver-docs-ddi #119: Do we still need?

rodwiddowson, 2018-06-11:
Back in Win7 we needed this to be set. See Christian Allred's comment here http://www.osronline.com/showThread.cfm?link=212938

Is this still true?  If so you'd better make the changes here.

PS where is this in github?  I was looking for it in windows-driver-docs so as to create a PR

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2336db2f-2258-6d12-3357-c1baf672b0d5
* Version Independent ID: f50e60b8-7d4a-9933-0d37-5d73609b9ee0
* Content: [_SRV_OPEN_ECP_CONTEXT](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/ns-ntifs-_srv_open_ecp_context)
* Content Source: [wdk-ddi-src/content/ntifs/ns-ntifs-_srv_open_ecp_context.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ns-ntifs-_srv_open_ecp_context.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

rodwiddowson, 2018-06-11:
The title got scrozzed and should read:
Do we still need `HKLM\SYSTEM\CurrentControlSet\Services\LanmanServer\Parameters\enableecp = 1`

EliotSeattle, 2018-06-11:
Good question, @rodwiddowson 
@aahill can you please investigate and update the topic?

rodwiddowson, 2018-06-13:
@aahill  While you are checking can you get the devs to comment on this post.  I have not done the experiment (yet) but it would indicate that SRV2 isn't setting the GUID.

aahill, 2018-06-13:
Hi @rodwiddowson, the devs looked into this; it doesn't seem that the registry key exists anymore, so you shouldn't need to worry about setting the registry key for current systems. Does that help?

aahill, 2018-06-14:
I should add that I've updated the documentation to reflect this. Should be visible soon.

rodwiddowson, 2018-06-14:
All the way back to windows 7?

aahill, 2018-06-14:
They took a look, and it doesn't look to be a requirement (or have usage) in Storport miniports going back to Windows 7. It’s a different story for scsiport miniports, but that's a deprecated port driver library.

aahill, 2018-06-14:
Gah! I realize that I responded to the wrong thread with that last comment. I'm not sure if it's all the way back to Windows 7. I'll look further into this for you.

aahill, 2018-07-10:
Hi @rodwiddowson, sincere apologies that I let this go for so long. It looks like it got removed in rs3, so not too long ago. Thanks for your patience, and please reach out if you need any documentation updates future.

rodwiddowson, 2018-07-10:
So a registry setting is required for win7, win8, win8.1, but it isn’t documented?

Also note that anecdotally it does t work for srv2, which needs documentation

aahill, 2018-07-10:
Thanks for the follow up!

I edited documentation for the registry key, including the windows version (Up to Windows 10, version 1709) and have it waiting to go live. Should be visible in a day or so.

I'm a bit confused about what you mean by srv2 - Is that a specific API, or is it a header file? Having some trouble finding it. Let me know and I'll look into it! Thanks!

rodwiddowson, 2018-07-11:
Thanks @aahill I hadn't realized it wasn't live yet.

About SRV2 - it is the replacement for SRV.SYS (which implements the SMB server in kernel and hence is responsible for sending the ECP).  Rumor on the stress is that when SVR2.SYS sends an MJ_CREATE it fails to garnish the IRP with this ECP.  I don't have access to the source & Devs but you do so I hope you'd confirm.
I have a UI to investigate this over the next month or two and if its true I'll open another case so you can clarify this in the doco (or get the stuff fixed).  If it isn't true I'll just pop a note in here,

Meanwhile I'll close this case.  Thanks very much again

closed 2018-07-11 by rodwiddowson