MicrosoftDocs/windows-driver-docs-ddi #121: Missing FILE_STAT_INFORMATION

rodwiddowson, 2018-06-12:
FILE_STAT_LX_INFORMATION is there (but with nothing in the Description Column).

Cross reference [Issue #74](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/issues/74)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0f39a405-8a1a-8d58-a796-05691153a17c
* Version Independent ID: bf989985-3b5f-0304-f20e-bc1a7a2838ae
* Content: [ntifs.h header](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/)
* Content Source: [wdk-ddi-src/content/ntifs/index.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/index.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

aahill, 2018-06-12:
Hi @rodwiddowson, looks like the FILE_STAT_LX_INFORMATION had a blank description attribute in the yaml block. I've updated it, so the description should show up within the next few days. Feel free to reopen this if you don't see it. Thanks!

closed 2018-06-12 by aahill

rodwiddowson, 2018-06-12:
@aahill thanks, but there was a second part.  There is a file info class FileStatInformation this came in in rs2 or rs3 but is not documented, neither is FILE_STAT_INFORMATION.

FileStatLxInformation which cropped up in rs4 is documented, and now it’s fully documented, but FileStatInformation isn’t (AFAICS)

Thanks