MicrosoftDocs/windows-driver-docs-ddi #131: Scott explains it all again

rodwiddowson, 2018-06-21:
https://www.osr.com/blog/2018/06/20/beware-mmbuildmdlfornonpagedpool-kernel-stacks/

Also note that code or data sections from executables cannot be used either.  You may want to add that to the documentation

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f6e8fd0c-7da6-be9d-11f4-84b284a71e98
* Version Independent ID: a4a96e43-35b6-a72f-926a-d1a2d35941a1
* Content: [MmBuildMdlForNonPagedPool function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-mmbuildmdlfornonpagedpool)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-mmbuildmdlfornonpagedpool.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-mmbuildmdlfornonpagedpool.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-21:
Thanks, @rodwiddowson 
@PageWriter-MSFT can you please look to see how best to incorporate and link to these suggested additions or file bugs? Thanks!!

PageWriter-MSFT, 2018-07-25:
Thanks for suggesting such a great post! I have updated the topic with information about the MDL:  must be either in non-paged pool, or in memory that is locked down (the physical pages cannot be traded). 

Changes should be live soon.

Page Writer

closed 2018-07-25 by PageWriter-MSFT