MicrosoftDocs/windows-driver-docs-ddi #136: _HW_INITIALIZATION_DATA isn't correct.

Ron-Hu, 2018-06-26:
Structure in 17134 Storport.w you must pull the correct header for documentation.  #ifdef &lt; Win8 is above here.  &gt;= Win8

#if (NTDDI_VERSION &gt;= NTDDI_WIN8)

//
// Define new HW_INITIALIZATION_DATA to support extended SRB type. Going
// forward, VIRTUAL_HW_INITIALIZATION is consolidated into
// HW_INITIALIZATION_DATA so there is only one structure to maintain.
//
typedef struct _HW_INITIALIZATION_DATA {

  ULONG                       HwInitializationDataSize;
  INTERFACE_TYPE              AdapterInterfaceType;
  PHW_INITIALIZE              HwInitialize;
  PHW_STARTIO                 HwStartIo;
  PHW_INTERRUPT               HwInterrupt;

  //
  // If miniport is physical, HwFindAdapter shall be of type PHW_FIND_ADAPTER.
  // If miniport is virtual, HwFindAdapter shall be of type
  // PVIRTUAL_HW_FIND_ADAPTER.
  //
  PVOID                       HwFindAdapter;
  PHW_RESET_BUS               HwResetBus;
  PHW_DMA_STARTED             HwDmaStarted;
  PHW_ADAPTER_STATE           HwAdapterState;
  ULONG                       DeviceExtensionSize;
  ULONG                       SpecificLuExtensionSize;
  ULONG                       SrbExtensionSize;
  ULONG                       NumberOfAccessRanges;
  PVOID                       Reserved;
  UCHAR                       MapBuffers;
  BOOLEAN                     NeedPhysicalAddresses;
  BOOLEAN                     TaggedQueuing;
  BOOLEAN                     AutoRequestSense;
  BOOLEAN                     MultipleRequestPerLu;
  BOOLEAN                     ReceiveEvent;
  USHORT                      VendorIdLength;
  PVOID                       VendorId;
  union {
    USHORT ReservedUshort;
    USHORT PortVersionFlags;
  } ;
  USHORT                      DeviceIdLength;
  PVOID                       DeviceId;
  PHW_ADAPTER_CONTROL         HwAdapterControl;
  PHW_BUILDIO                 HwBuildIo;

  //
  // Following 5 callbacks are virtual miniport specific and should not be set
  // by physical miniports.
  //
  PHW_FREE_ADAPTER_RESOURCES  HwFreeAdapterResources;
  PHW_PROCESS_SERVICE_REQUEST HwProcessServiceRequest;
  PHW_COMPLETE_SERVICE_IRP    HwCompleteServiceIrp;
  PHW_INITIALIZE_TRACING      HwInitializeTracing;
  PHW_CLEANUP_TRACING         HwCleanupTracing;

  //
  // New fields should only be added after this one as the previous fields match
  // the HW_INITIALIZATION definition for pre-Win8.
  //
  PHW_TRACING_ENABLED         HwTracingEnabled;

  // Used to indicate features supported by miniport such as virtual miniports
  ULONG             FeatureSupport;

  // New fields added to support extended SRB format(s)
  ULONG             SrbTypeFlags;
  ULONG             AddressTypeFlags;
  ULONG             Reserved1;

  //
  // Unit control callback
  //
  PHW_UNIT_CONTROL  HwUnitControl;

} HW_INITIALIZATION_DATA, *PHW_INITIALIZATION_DATA;

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3eb29277-0674-6534-c5b3-33f23c8d1139
* Version Independent ID: 56fc2963-d527-21e9-5411-8d721ce1bc81
* Content: [_HW_INITIALIZATION_DATA](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/storport/ns-storport-_hw_initialization_data#feedback)
* Content Source: [wdk-ddi-src/content/storport/ns-storport-_hw_initialization_data.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/storport/ns-storport-_hw_initialization_data.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-26:
Thanks, @Ron-Hu 
@aahill can you triage and correct, please?

aahill, 2018-07-03:
Thanks for your help and patience Ron, I was able to find the right header and location. I've updated the page with the correct information, but the structure's syntax block is causing some strange behavior in our content rendering system. I'm following up with our support team, and will update the thread once we have the issue figured out, and the update live.

aahill, 2018-07-19:
Hi again @Ron-Hu, the rendering system should be fixed soon, after which I'll promote the changes live. Should be a day or so. 

Thanks again for bringing this to our attention!

closed 2018-07-19 by aahill