MicrosoftDocs/windows-driver-docs-ddi #145: Example

Ma-XX-oN, 2018-07-06:
An example would be nice.  Unsure what `BackTraceHash` is, and how it is used.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9f961a72-1c65-afde-5aca-3d567b96cd6c
* Version Independent ID: 1b8e9d4c-b7ed-e507-fa1f-bd9ef3c2be38
* Content: [RtlCaptureStackBackTrace function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-rtlcapturestackbacktrace)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-rtlcapturestackbacktrace.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-rtlcapturestackbacktrace.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

lorihollasch, 2019-05-22:
@Ma-XX-oN , thanks for submitting this issue and HUGE apologies that it took us so long to get to it!!! I have updated the page with additional information and hopefully some clearer language. This hash is useful like other hashes in CS algorithms.

lorihollasch, 2019-05-22:
@Ma-XX-oN , looks like my comment didn't take the first time so trying again. Thanks for submitting this issue, and HUGE apologies that it took us so long to get to it. I have added additional info and clarifying language to the page to help make things more clear. BackStackHash can be used like other hashes in CS algorithms - if it is useful to you.

closed 2019-05-22 by lorihollasch