MicrosoftDocs/windows-driver-docs-ddi #146: REPARSE_POINT, OPLOCK

SIISII, 2018-07-09:
I think it is necessary to add a link to a description what REPARSE POINT and OPLOCK are and how they can be used.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d810c5ee-d635-7303-8643-01a03e9a81c2
* Version Independent ID: 1e0ce8da-ff20-752a-42ea-6a5bf180190d
* Content: [ZwCreateFile function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-zwcreatefile#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-zwcreatefile.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-zwcreatefile.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-09:
Thanks for the suggestion, @SIISII 
@PageWriter-MSFT can you please triage this request and add it to your backlog as appropriate?

PageWriter-MSFT, 2018-07-16:
I have added these links to the topic. The topic should be live soon.

 [Reparse Points](https://docs.microsoft.com/en-us/windows/desktop/FileIO/reparse-points)

[Opportunistic Locks](https://msdn.microsoft.com/en-us/library/windows/desktop/aa365433(v=vs.85).aspx)

Thanks @SIISII 

Page Writer

closed 2018-07-16 by PageWriter-MSFT