MicrosoftDocs/windows-driver-docs-ddi #149: Does IRQL respect the ExecutionLevel of the queue?

jakeoshins, 2018-07-10:
This page should explicitly state whether the IRQL of this callback is adjusted by the ExecutionLevel parameter of the WDF_OBJECT_ATTRIBUTES supplied at time of queue creation.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 852d986e-17ed-8ac1-6783-1a74aee7f043
* Version Independent ID: 061518a3-b18b-d469-06f6-4e4d59aad210
* Content: [WdfIoQueueReadyNotify function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdfio/nf-wdfio-wdfioqueuereadynotify)
* Content Source: [wdk-ddi-src/content/wdfio/nf-wdfio-wdfioqueuereadynotify.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfio/nf-wdfio-wdfioqueuereadynotify.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

Shyamal-V, 2018-07-11:
Yes the execution level is respected. This should be clarified in WdfIoQueueReadyNotify, WdfIoQueueStop, WdfIoQueueDrain, WdfIoQueuePurge.

tedhudek, 2018-07-12:
These changes are live.  See extra note on https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdfio/nc-wdfio-evt_wdf_io_queue_state and pointers to it on WdfIoQueueReadyNotify, WdfIoQueueStop, WdfIoQueueDrain, WdfIoQueuePurge.

closed 2018-07-12 by tedhudek