MicrosoftDocs/windows-driver-docs-ddi #152: The listing of hash algorithms (BDCB_IMAGE_INFORMATION::ImageHashAlgorithm) lacks descriptions

cyren-oschneider, 2018-07-12:
`BDCB_IMAGE_INFORMATION::ImageHashAlgorithm` is currently described by giving a listing of possible values. Alas, these values are only defined in `WinCrypt.h` (which is a user-mode header) and it's not clear what their meaning is.

By digging a bit into the user mode header and getting to its documentation by way of some of the function names declared there, I ended up finding [`ALG_ID`](https://docs.microsoft.com/en-us/windows/desktop/seccrypto/alg-id) which seems to be a superset of the identifiers listed.

And indeed when looking at the ELAM sample [README.md](https://github.com/Microsoft/Windows-driver-samples/blob/master/security/elam/README.md) we can find:

    ElamSample: Certificate thumb print algorithm = 0x0000800c.

... and sure enough we can find `CALG_SHA_256` listed as value `0x0000800c` (which matches the expectations for the ELAM).

It would be nice if that connection could be made and either the relevant subset (i.e. hash algorithms) could be taken over into the `BDCB_IMAGE_INFORMATION` documentation or at least a remark could be added to that end or a link to the `ALG_ID` article could be provided in **See also**.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4080585d-6e46-b6c7-115d-dfa97477eb33
* Version Independent ID: 181e5da7-4032-e667-31af-14460bf72d4a
* Content: [_BDCB_IMAGE_INFORMATION](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/ns-ntddk-_bdcb_image_information)
* Content Source: [wdk-ddi-src/content/ntddk/ns-ntddk-_bdcb_image_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/ns-ntddk-_bdcb_image_information.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

cyren-oschneider, 2018-07-12:
By the way, I failed to mention that the identifiers ("defines") mentioned in that list without descriptions are generally unavailable in kernel mode. But perhaps that was obvious. Perhaps if one went and selectively copied the appropriate parts from `WinCrypt.h` (see at the bottom of this comment). Across the range of WDKs/SDKs I have, none seems to have a kernel mode header defining (PCRE) `C?ALG_(SID_)?SHA_(256|384|512)`, for example:

```
TOTAL:    30 matches in 6 files  (11593 other files without matches are not listed)
1 match in C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\um\Mprapi.h
    1806      DWORD                       certAlgorithm; // Should always be CALG_SHA_256
9 matches in C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\um\wincrypt.h
     254  #define ALG_SID_SHA_256                 12
     255  #define ALG_SID_SHA_384                 13
     256  #define ALG_SID_SHA_512                 14
     330  #define CALG_SHA_256            (ALG_CLASS_HASH | ALG_TYPE_ANY | ALG_SID_SHA_256)
     331  #define CALG_SHA_384            (ALG_CLASS_HASH | ALG_TYPE_ANY | ALG_SID_SHA_384)
     332  #define CALG_SHA_512            (ALG_CLASS_HASH | ALG_TYPE_ANY | ALG_SID_SHA_512)
1 match in C:\Program Files (x86)\Windows Kits\10\Include\10.0.16299.0\um\Mprapi.h
    1845      DWORD                       certAlgorithm; // Should always be CALG_SHA_256
9 matches in C:\Program Files (x86)\Windows Kits\10\Include\10.0.16299.0\um\wincrypt.h
     266  #define ALG_SID_SHA_256                 12
     267  #define ALG_SID_SHA_384                 13
     268  #define ALG_SID_SHA_512                 14
     342  #define CALG_SHA_256            (ALG_CLASS_HASH | ALG_TYPE_ANY | ALG_SID_SHA_256)
     343  #define CALG_SHA_384            (ALG_CLASS_HASH | ALG_TYPE_ANY | ALG_SID_SHA_384)
     344  #define CALG_SHA_512            (ALG_CLASS_HASH | ALG_TYPE_ANY | ALG_SID_SHA_512)
1 match in C:\Program Files (x86)\Windows Kits\10\Include\10.0.17134.0\um\Mprapi.h
    1845      DWORD                       certAlgorithm; // Should always be CALG_SHA_256
9 matches in C:\Program Files (x86)\Windows Kits\10\Include\10.0.17134.0\um\wincrypt.h
     266  #define ALG_SID_SHA_256                 12
     267  #define ALG_SID_SHA_384                 13
     268  #define ALG_SID_SHA_512                 14
     342  #define CALG_SHA_256            (ALG_CLASS_HASH | ALG_TYPE_ANY | ALG_SID_SHA_256)
     343  #define CALG_SHA_384            (ALG_CLASS_HASH | ALG_TYPE_ANY | ALG_SID_SHA_384)
     344  #define CALG_SHA_512            (ALG_CLASS_HASH | ALG_TYPE_ANY | ALG_SID_SHA_512)
```

Those "`ALG_ID` crackers" from WinCrypt.h would also be generally useful for the same reason:

```
// ALG_ID crackers
#define GET_ALG_CLASS(x)                (x & (7 << 13))
#define GET_ALG_TYPE(x)                 (x & (15 << 9))
#define GET_ALG_SID(x)                  (x & (511))
```

A minimal header for kernel mode might look like this:

```
//
// Algorithm IDs and Flags
//

// ALG_ID crackers
#define GET_ALG_CLASS(x)                (x & (7 << 13))
#define GET_ALG_TYPE(x)                 (x & (15 << 9))
#define GET_ALG_SID(x)                  (x & (511))

// Algorithm classes
#define ALG_CLASS_HASH                  (4 << 13)

// Algorithm types
#define ALG_TYPE_ANY                    (0)

// Hash sub ids
#if (NTDDI_VERSION > NTDDI_WINXPSP2)
#define ALG_SID_SHA_256                 12
#define ALG_SID_SHA_384                 13
#define ALG_SID_SHA_512                 14
#endif //(NTDDI_VERSION > NTDDI_WINXPSP2)

// algorithm identifier definitions
#if (NTDDI_VERSION > NTDDI_WINXPSP2)
#define CALG_SHA_256            (ALG_CLASS_HASH | ALG_TYPE_ANY | ALG_SID_SHA_256)
#define CALG_SHA_384            (ALG_CLASS_HASH | ALG_TYPE_ANY | ALG_SID_SHA_384)
#define CALG_SHA_512            (ALG_CLASS_HASH | ALG_TYPE_ANY | ALG_SID_SHA_512)
#endif //(NTDDI_VERSION > NTDDI_WINXPSP2)
```

That's also roughly what I'll be using now, but guarding against the possibility that these values have been previously defined.

EliotSeattle, 2018-07-12:
Wow, THANKS, @cyren-oschneider 
@DOMARS let's you and I get together to talk about how we proceed here.

EliotSeattle, 2019-01-09:
@tedhudek 
I am embarrassed to say that I sat on this issue for so long, especially since Oliver gave us such amazing info in it. Can you take this and add it to the content after it's verified? 
Thanks,
@EliotSeattle

cyren-oschneider, 2019-01-11:
A happy new year to you and yours, first of all!

Since I don't know the exact roles of you folks, I just wanted to reiterate that this may not only be a documentation issue, given that a header for kernel mode may be a desirable outcome here as well. Just saying.

Best regards,

Oliver

tedhudek, 2019-05-20:
Hi @ghost apologies for the delay.  I've added a link from https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/ns-ntddk-_bdcb_image_information to the ALG_ID page (will go live later today) but I'm afraid that's all I'm equipped to do at the moment.  If you feel that additional doc updates are warranted here, you can always click edit from the page and open a PR.

closed 2019-05-20 by tedhudek