MicrosoftDocs/windows-driver-docs-ddi #155: nvme.h contains the following programming interfaces:

csm10495, 2018-07-14:
On the website I see:
nvme.h contains the following programming interfaces:

Though there are no interfaces or really anything listed. I also don't see that no the github source. I think we're missing documentation.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5d64f03a-9abd-cd29-b116-6d22b0f676d3
* Version Independent ID: 9186dc46-5348-9193-68d6-55901873db9f
* Content: [nvme.h header](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/nvme/index)
* Content Source: [wdk-ddi-src/content/nvme/index.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/nvme/index.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-16:
Thanks, @csm10495 
@PageWriter-MSFT can you look into why the elements in this header are not getting generated?

PageWriter-MSFT, 2018-07-16:
@csm10495 

This is a bug in our system. This should get updated soon. Also, this header hasn't been documented historically. I will file a ticket to get this header documented.

Thanks!

Page Writer

closed 2018-07-16 by PageWriter-MSFT